hku-mars/mlcc #2: Question about vpnp cost function

RoseWillow, 2021-11-12:
Dear author, thanks for your excellent work, but I am puzzled by the vpnp cost function, I did not understand why we need the line direction to compute the matrix V, and the physical meaning of this formula VRV.transpose
[image]
Look forward to your reply

ChongjianYUAN, 2021-11-16:
> Dear author, thanks for your excellent work, but I am puzzled by the vpnp cost function, I did not understand why we need the line direction to compute the matrix V, and the physical meaning of this formula VRV.transpose [image] Look forward to your reply

[image]
the line direction can help us build point-to-line residual but not point-to-point residual. Besides, the correct formulation of the point-to-line residual is shown in this picture. There are some minor bugs in the code, but they don't matter much after testing.

RoseWillow, 2021-11-16:
> [image] the line direction can help us build point-to-line residual but not point-to-point residual. Besides, the correct formulation of the point-to-line residual is shown in this picture. There [...]

Thank you for your reply, but I still don't know how to build point-to-line residuals. Can you tell me the source of the picture in your last reply?

ChongjianYUAN, 2021-11-16:
> Thank you for your reply, but I still don't know how to build point-to-line residuals. Can you tell me the source of the picture in your last reply?

Sorry, the content in this picture is written by myself, so there is no source. You can refer to the vector projection matrix, then you will know how to build point-to-line residuals.

closed 2021-11-16 by samsdolphin

RoseWillow, 2021-11-16:
> Sorry, the content in this picture is written by myself, so there is no source. You can refer to the vector projection matrix, then you will know how to build point-to-line residuals.

OK，thank you very much!