hku-mars/mlcc #3: calibration of one camera and one lidar

zzkslam, 2021-12-05:
Thanks for your excellent work. I want to use the code to calibrate one camera and one lidar. Can you give me some advice about how to change the code?

samsdolphin, 2021-12-07:
Hi zzkslam, for single LiDAR and single-camera extrinsic calibration, we refer you to our previous work https://github.com/hku-mars/livox_camera_calib. Thanks~

closed 2021-12-07 by samsdolphin

jdtsui, 2022-02-05:
> Hi zzkslam, for single LiDAR and single-camera extrinsic calibration, we refer you to our previous work https://github.com/hku-mars/livox_camera_calib. Thanks~

Thanks for your excellent work, too. But one question here: Have you updated the new algorithm (using adaptive voxelization to extract lines) to replace the old one (using RANSAC only) in "livox_camera_calib"?

zying0393, 2022-07-15:
> 

Same question. Looking forward to your reply!

zying0393, 2022-07-15:
> Thanks for your excellent work, too. But one question here: Have you updated the new algorithm (using adaptive voxelization to extract lines) to replace the old one (using RANSAC only) in [...]

Same question. Looking forward to your reply!

af-doom, 2023-04-14:
> Same question. Looking forward to your reply!

Have you solved this problem