hku-mars/mlcc #6: Error for calib_camera.launch

JangChangWon, 2022-02-25:
hello! Thanks for your excellent works !!

When I follow the guide, I run roslaunch mlcc calib_camera.launch, some errors occur, below are some msg.
my env is ubuntu18.04, ros is melodic.

> [ INFO] [1645761068.171630692]: Camera 0 Configuration
> Camera Matrix: 
> [863.590518437255, 0.116814496662654, 621.666074631063;
>  0, 863.100180533059, 533.971978652819;
>  0, 0, 1]
> Distortion Coeffs: 
> [-0.0944205499243979;
>  0.09467276777765039;
>  -0.00807970960613932;
>  8.07461209775283e-05;
>  6.527166468758789e-05]
> Extrinsic Params: 
> [0, -1, 0, 0;
>  0, 0, -1, 0;
>  1, 0, 0, 0;
>  0, 0, 0, 1]
> Rotation error:0
> [ INFO] [1645761068.172541533]: Camera 1 Configuration
> Camera Matrix: 
> [863.081687640302, 0.176140650303666, 628.941349825503;
>  0, 862.563371991533, 533.00290953509;
>  0, 0, 1]
> Distortion Coeffs: 
> [-0.0943795554942897;
>  0.09829982415249131;
>  -0.0125418048527694;
>  0.000456235380677041;
>  -8.73113795357082e-05]
> Extrinsic Params: 
> [0, -1, 0, 0;
>  0, 0, -1, 0;
>  1, 0, 0, 0;
>  0, 0, 0, 1]
> Rotation error:0
> total ext_number:3
> [calib_camera-2] process has died [pid 7304, exit code -9, cmd /home/A/ros_ws/catkin_ws_mlcc/devel/lib/mlcc/calib_camera /home/A/ros_ws/catkin_ws_mlcc/src/mlcc/config/left.yaml /home/A/ros_ws/catkin_ws_mlcc/src/mlcc/config/right.yaml /home/A/ros_ws/catkin_ws_mlcc/src/mlcc/config/avia_stereo.yaml /home/A/ros_ws/catkin_ws_mlcc/src/mlcc/result __name:=calib_camera __log:=/home/A/.ros/log/29b15922-95ee-11ec-ad7f-a8a15920a9d6/calib_camera-2.log].
> log file: /home/A/.ros/log/29b15922-95ee-11ec-ad7f-a8a15920a9d6/calib_camera-2*.log

Is there any config or setting that I need to change? thanks for your reply.
Thanks

JangChangWon, 2022-02-25:
it is just path problem

closed 2022-02-25 by JangChangWon

GarethZhang, 2022-04-26:
Hi @JangChangWon,

I am encountering the same error. Could you please elaborate on what path problem you had? There seems to be no error related to the launch file?

Thank you!

93won, 2022-04-29:
> it is just path problem

I have a same issue. What path did you change?

JangChangWon, 2022-05-01:
> > it is just path problem
> 
> I have a same issue. What path did you change?

JangChangWon, 2022-05-01:
@93won 
Hi,
I just change the path of "LiDARFilesPath" on "/mlcc/config/avia_stereo.yaml"

93won, 2022-05-22:
> @93won Hi, I just change the path of "LiDARFilesPath" on "/mlcc/config/avia_stereo.yaml"

Thank you so much! I solved the problem :)