hku-mars/mlcc #8: quastion about your voxel: does it only for livox lidar:

mingkunwxc, 2022-03-02:
Dear authors:
In your project,I found that loc_xyz sames like must more than 0.
And,i found the livox lidar's frame is like this,x>0,y>0,z>0. 
but ,I want using it for robosens lidar .
If I need modify this param? 
 thanks for your help.

_________________________________________
      for (int j = 0; j < 3; j++)
		{
			loc_xyz[j] = pt_trans[j] / voxel_size;
			if (loc_xyz[j] < 0)
				loc_xyz[j] -= 1.0;
		}
___________________________________________________

samsdolphin, 2022-03-04:
Hi mingkunwxc, this code works for mechanical spinning LiDAR like RoboSense, Velodyne, etc. The code you included here only means if the coordinate of the transformed point is negative, it should be put into a voxel with a negative location index.

mingkunwxc, 2022-04-07:
> Hi mingkunwxc, this code works for mechanical spinning LiDAR like RoboSense, Velodyne, etc. The code you included here only means if the coordinate of the transformed point is negative, it should be put into a voxel with a negative location index.

thanks, I got it

closed 2022-05-10 by samsdolphin