hku-mars/mlcc #15: Pose_refine reported the following error: Assertion `!std::isnan(residual2)'

Yurri-hub, 2022-06-19:
Thanks for your excellent work. 
When I converted the KITTI360 data into the required format for pose optimization, the problem of "Assertion `!std::isnan(residual2)`" was reported
After I conducted a series of tests, I found that when the number of frames I used was less than or equal to 35, there was no problem with Pose_refine (0.pcd~34.pcd). But as long as the number of frames is greater than 35, the error will be reported.
```
process[pose_refine-1]: started with pid [26669]
process[rviz-2]: started with pid [26676]
---------------------
iteration 0
pose_refine: /home/ly/catkin_ws/src/mlcc/include/pose_refine.hpp:272: void LM_OPTIMIZER::optimize(): Assertion `!std::isnan(residual2)' faile
d.
[pose_refine-1] process has died [pid 26669, exit code -6, cmd /home/ly/catkin_ws/devel/lib/mlcc/pose_refine __name:=pose_refine __log:=/home
/ly/.ros/log/f4dda018-eeec-11ec-a53c-04d4c45d3faf/pose_refine-1.log].
log file: /home/ly/.ros/log/f4dda018-eeec-11ec-a53c-04d4c45d3faf/pose_refine-1*.log
```

Yurri-hub, 2022-06-20:
The reason is that the format of the initial pose is wrong

closed 2022-06-20 by Yurri-hub

19855179496, 2024-03-15:
> The reason is that the format of the initial pose is wrong原因是初始姿势的格式不对

what does it mean?