hku-mars/mlcc #18: NAN observed after solving AX=B, leading to failure when optimizing extrinsics

MiBei0325, 2023-02-22:
Hello, I ran into a problem with my own dataset that `matX` contains NAN elements after solving linear equation AX=B using `cv::solve()` function in `extrinsic_refine.hpp`, causing failure in optimizing extrinsic between base LiDAR and ref LiDAR. 

I've checked and made sure that my input point cloud does not contain NAN elements, what are possible solution to this problem to keep the optimization process running, thank you!

samsdolphin, 2023-02-23:
Hi @MiBei0325, have you tried using different `voxel sizes` and `eigen_ratio`?
Please let me know if this issue remains. I plan to update the project code but I am pretty busy this week.

samsdolphin, 2023-02-28:
Hi @Nicky0325, I have updated the code. Please let me know if your problem is solved.

Nicky0325, 2023-04-04:
Thank you, this is exactly the case. I found that the feature map was not correct due to faulty configuration of these two parameters. Thank you for the hints and the update, much appreciation!

closed 2023-04-04 by Nicky0325