hku-mars/mlcc #20: Does it support calibration of a small FoV lidar and a spinning (360 horizontal FoV) Lidar?

hiyyg, 2023-04-21:


samsdolphin, 2023-04-21:
Yes. We tested on 1 Ouster (with 360 FoV) LiDAR and 4 HAP LiDAR (with 120 FoV). Just make sure you rotate and hold the whole sensor suite several times (check our video) such that there is some FoV overlap between two adjacent poses for the small FoV LiDAR. Hope this helps.

hiyyg, 2023-04-21:
Thanks. Does it work well for other types of moving, such as a car mounted with the different lidars moves near some walls? Because a rotating platform is not avilable.

samsdolphin, 2023-04-24:
> Thanks. Does it work well for other types of moving, such as a car mounted with the different lidars moves near some walls? Because a rotating platform is not avilable.

Yes, the above sensor setup was tested on a vehicle platform. We drove the car following a circular trajectory to collect the data.

hiyyg, 2023-04-25:
Could you give some advises about the requirements for the calibration environment and the driving trajectory for a vehicle? It would nice if you could post some pictures for this. Thanks!