hku-mars/mlcc #23: The PCD obtained by velodyne16 does not seem to yield good voxels

af-doom, 2023-07-11:
[image]

samsdolphin, 2023-09-11:
Hi @af-doom, I apologize for my late reply. Have you solved this issue? Can you share the point cloud, image, and rosbag so I can help you to debug it?

af-doom, 2023-09-12:
Thank you very much for your reply. The image, bag, and PCD data are attached. Thank you very much
test_data.zip

samsdolphin, 2023-09-15:
> Thank you very much for your reply. The image, bag, and PCD data are attached. Thank you very much test_data.zip
> […](#)

Hi @af-doom, I can't find your attached files here. Can you double-check?

af-doom, 2023-09-15:
I'm very sorry, could you try this one

https://chdeducn-my.sharepoint.com/:u:/g/personal/2018901230_chd_edu_cn/EdmgF7ZRwPZClsj9YYD04noBr5z5VAaVb_P0remBl7ZrZw?e=X3SjHY

lsangreg, 2023-11-24:
Hi, how did you resolve?

Chatoyant19, 2024-01-24:
I also have this problem.Did you solve it?

lsangreg, 2024-01-25:
@Chatoyant19  tbh i wasn't able to reproduce their results  with their data, what about you?

Chatoyant19, 2024-01-26:
> @Chatoyant19 tbh i wasn't able to reproduce their results with their data, what about you?

I can get a good result with their data.I think it is necessary  to adjust some parameters, when using myself datas. Because after I changed 'voxel_size' and 'eigen_ratio', I can extract more useful planes.But it is not enough, for example, the same wall cannot be extracted completely when the plane is extracted.

lsangreg, 2024-01-26:
> I can get a good result with their data.I think it is necessary to adjust some parameters, when using myself datas. Because after I changed 'voxel_size' and 'eigen_ratio', I can extract more useful [...]

I think at least it is important to reproduce what they say " the average translation and rotation errors down to 6mm and 0.09 degrees for LiDAR-camera" , which for me is more or less angular error 0.067 degree baseline error 0.025 m ( should be 0.006).

With this being said I will also try to investigate more eigen and voxel size... I was working more on the edge detection intensity as my data are synthetic and highly defined...

Can I ask you which did error you introduce to your initial extrinsics?

lsangreg, 2024-02-16:
@Chatoyant19  did you obtain better results?

Chatoyant19, 2024-08-06:
> @Chatoyant19 did you obtain better results?

No, I didn't. I know that my problem lies in extracting planar features from the LiDAR point clouds. I 'm unable to extract enough and high-quality planar features.Maybe use PCA to extract planar features is closely related to the quality of point cloud mapping? My LiDAR model is the Hesai FT120, which has relatively poor ranging accuracy.

af-doom, 2024-08-06:
> No, I didn't. I know that my problem lies in extracting planar features from the LiDAR point clouds. I 'm unable to extract enough and high-quality planar features.Maybe use PCA to extract planar [...]

Hello, do you have data about FT120 and camera? Can you share it with me for testing?

Chatoyant19, 2024-08-07:
> Hello, do you have data about FT120 and camera? Can you share it with me for testing?

Of course, can I have your email address? I'll send you the relevant data.

af-doom, 2024-08-07:
***@***.*** This is my email address, thank you

Chatoyant19, 2024-08-23:
> […](#)

Sorry, the email information has been encrypted, I cannot access it.