hku-mars/mlcc #28: MID360 and camera calibration

af-doom, 2024-01-02:
[image]
Hello, thank you very much for your open source project, I am not able to get good results when calibrating a single lidar and camera, the alignment results are shown in the picture. How should I adjust the parameters? The alignment works well when testing the data set you posted. Do I need to change other parameters for the pinhole camera model?

lsangreg, 2024-01-08:
Hi, did you solve your problem?

zhangyuran-gg, 2024-10-17:
> ![2024-01-02 16-09-41 的屏幕截图](https://private-user-images.githubusercontent.com/91470350/293649854-92cc1eb4-cdc6-4c5d-976b-a886addc3cbf.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjkxNDc5MTAsIm5iZiI6MTcyOTE0NzYxMCwicGF0aCI6Ii85MTQ3MDM1MC8yOTM2NDk4NTQtOTJjYzFlYjQtY2RjNi00YzVkLTk3NmItYTg4NmFkZGMzY2JmLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMTclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDE3VDA2NDY1MFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTM4MDkwY2IyNzEwNDA3OTZmY2MwZWQ1OGQ0YWI1YjMwOGI1ZDZjMGMwYTJhMGU2YWRkMTg0ZmZiZmRhZjQzM2QmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.9xo826GzeK1QcDarkbbe6ZRFcjdLvq_0ypGiiJ8fFJA)您好，非常感谢您的开源项目，我在校准单个激光雷达和相机时无法获得很好的结果，对准结果如图所示。我应该如何调整参数？在测试您发布的数据集时，对齐效果很好。我需要更改针孔相机型号的其他参数吗？
> 
> 我的数据链接 ： https://1drv.ms/u/s!AuQ1Ilp0ikW-i3H-ZFp6POdNKQST?e=gLYFPf

你好，你解决你的问题了吗